RAKWireless/RAK-STM32-RUI #1: Serial config not modified.

IvanAmg, 2022-07-29:
Hi!
I'm currently using the RAK3172 module for my project.
My application needs to have an even parity mode in order to read and send data through Serial.
For this communication, I'm using USART1, which is Serial1 according to the documentation.
Tried first with Serial1.begin(2400, SERIAL_8E1); with no success.
I have tried, also,  modifying the core of RUI to select a default mode of SERIAL_8E1 but it seems like the Serial won't change in any mode.
If my code is needed, I'm able to share it.
Thanks!

beegee-tokyo, 2022-07-31:
Hello @IvanAmg 

It is a know bug and will be fixed in the next release.

closed 2022-07-31 by beegee-tokyo

sabas1080, 2022-09-13:
@beegee-tokyo any update? Thanks