RAKWireless/RAK-STM32-RUI #5: dtostrf() build failure

lyusupov, 2022-09-25:
RUI3 SDK release 3.4.2
WisDuo_RAK3172_Evaluation_Board

Sketch _dtostrf_test.ino_:

```
#include <avr/dtostrf.h>

float f_val = 123.6794;
char outstr[15];

void setup()
{
  dtostrf(f_val,7, 3, outstr);
}
void loop()
{
}
```

Build:

```
$ arduino  --verify --verbose-build dtostrf_test.ino

< ... skipped ... >

system/WisDuo_RAK3172_Evaluation_Board/librui_v3_api.a(dtostrf.o): in function `dtostrf':
(.text.dtostrf+0x14): undefined reference to `asm'
collect2: error: ld returned 1 exit status
exit status 1

```

lyusupov, 2022-09-27:
The issue is confirmed with recent RUI3 SDK release [v3.5.2b_175](https://github.com/RAKWireless/RAK-STM32-RUI/releases/tag/v3.5.2b_175) as well.

beegee-tokyo, 2024-01-22:
Fixed in V4.1.0

closed 2024-01-22 by beegee-tokyo